efabless/clear #14: TapeOut: 2020-07 Unused power signal mismatch

d-m-bailey, 2022-07-12:
This issue needs to be resolved or waived before tapeout.

```
https://gitlab.com/api/v4/projects/36768579/repository/files/tapeout%2Foutputs%2Foas%2Fcaravel_2204289a.oas/raw?ref=main
15adb4f6dd8e482041f8bad67a281c75c6fdd0cd  caravel_2204289a.oas
```

Some of the power nets that are unused in `user_project_wrapper` do not match the connections to `caravel` `chip_io`.
| chip_io | user_project_wrapper |
| :---: | :---: |
| vssa1_core | vssd2 |
| vssd2_core | vdda1 |
| vdda1_core | vdda2 |
| vdda2_core | vssa1 |

These nets are not actually used in user_project_wrapper, so relabeling would be sufficient.

[image]

mkkassem, 2022-07-25:
We need to check if it impacts other projects.

marwaneltoukhy, 2022-08-15:
@d-m-bailey I was looking at the pdn and looks like you are highlighting the wrong net, ``vssd2`` is the 5th ring from the right, not the third
[image]