efabless/clear #26: What is the current situation of clear eFPGA macro?

WazaAbdulkadir, 2023-05-29:
I have been seing some commits to develop clear eFPGA since mid-March. 
What is the current situation of clear eFPGA project?
I saw some issues are not answered, is everything suitable for use?

Any information would be very useful.

mo-hosni, 2023-05-30:
Clear was updated based on new updates in SOFA. 
The HDL updates can be found here: https://github.com/lnis-uofu/SOFA/tree/master/SOFA_A/FPGA88_SOFA_A

Can you elaborate more on the issues you're seeing?

WazaAbdulkadir, 2023-05-31:
Actually I want to learn things like;  
 
- Are there people to answer issues, since I will try to generate clear efpga from stratch.
- What is the recent effort on GH for? 
- Are gds files in  the repo ready to be tapeout?  
- Is product on sale?  
 
Just before starting to work on this repo, I wanted to know overall situation.

WazaAbdulkadir, 2023-06-05:
What does make Clear eFPGA especially "eFPGA" ?  Is it because it's size is small? Or there are other special reasons to call Clear FPGA an eFPGA.

Martoni, 2024-01-31:
@WazaAbdulkadir 

> What does make Clear eFPGA especially "eFPGA" ? Is it because it's size is small? Or there are other special reasons to call Clear FPGA an eFPGA.

The 'e' of **e**FPGA mean «embedded». Then we call it eFPGA because it's embedded in [caravel](https://github.com/efabless/caravel) SoM I think.