efabless/clear #34: Toolchain for firmware development

benjaminSchilling33, 2024-07-24:
Hi all, 

first: Congrats on achieving your goal of providing a fully open source FPGA!
I ordered my Clear board in February 2022 and finally received it two weeks ago.
Thanks for your great work!

Now I'm wondering how to get firmware on the device.
The "firmware" section in the README is a good starting point but up to now does not provide any details on the build dependencies required to build the firmware.

I assume that the documentation [here](https://caravel-harness.readthedocs.io/en/latest/programming.html) and it's references are sufficient or is there anything needed in addition?

If you can give me some initial directions, I'm happy to contribute the basic documentation after getting the examples to run.

Greetings from Germany!
Benjamin

Martoni, 2024-07-25:
I managed to compile blinking led example from [here](https://github.com/efabless/clear/tree/main/firmware/blink) .

I'm on debian like system:

```bash
$ sudo apt install gcc-riscv64-unknown-elf
$ make
generating hex for [blink]
riscv64-unknown-elf-gcc -I../fpga_bitstreams/ -I../common/ -O0 -mabi=ilp32 -march=rv32i -D__vexriscv__ -Wl,-Bstatic,-T,../common/sections.lds,--strip-debug -ffreestanding -nostdlib -o blink.elf ../common/crt0_vex.S ../common/isr.c ../common/gpio_program.c blink.c
riscv64-unknown-elf-objdump -s  blink.elf > blink.lst
riscv64-unknown-elf-objcopy -O verilog blink.elf blink.hex
sed -ie 's/@10/@00/g' blink.hex
$ python3 -m pip install pyftdi
$ make flash
generating hex for [blink]
python3 ../util/ftdi_flash.py blink.hex
Success: Found one matching FTDI device at ftdi://ftdi:232h:1:1c/1
 
Resetting Flash...
status = 0x00
 
JEDEC = b'ef4016'
Erasing chip...
done
status = 0x0
setting address to 0x0
addr 0x0: flash page write successful
addr 0x100: flash page write successful
addr 0x200: flash page write successful
addr 0x300: flash page write successful
addr 0x400: flash page write successful
addr 0x500: flash page write successful
addr 0x600: flash page write successful
addr 0x700: flash page write successful
addr 0x800: flash page write successful
addr 0x900: flash page write successful
addr 0xa00: flash page write successful
addr 0xb00: flash page write successful
addr 0xc00: flash page write successful
addr 0xd00: flash page write successful
addr 0xe00: flash page write successful
addr 0xf00: flash page write successful
addr 0x1000: flash page write successful
addr 0x1100: flash page write successful
addr 0x1200: flash page write successful
addr 0x1300: flash page write successful
addr 0x1400: flash page write successful

total_bytes = 5376
status reg_1 = 0x0
status reg_2 = 0x2
************************************
verifying...
************************************
status reg_1 = 0x0
status reg_2 = 0x2
setting address to 0x0
addr 0x0: read compare successful
addr 0x100: read compare successful
addr 0x200: read compare successful
addr 0x300: read compare successful
addr 0x400: read compare successful
addr 0x500: read compare successful
addr 0x600: read compare successful
addr 0x700: read compare successful
addr 0x800: read compare successful
addr 0x900: read compare successful
addr 0xa00: read compare successful
addr 0xb00: read compare successful
addr 0xc00: read compare successful
addr 0xd00: read compare successful
addr 0xe00: read compare successful
addr 0xf00: read compare successful
addr 0x1000: read compare successful
addr 0x1100: read compare successful
addr 0x1200: read compare successful
addr 0x1300: read compare successful
addr 0x1400: read compare successful

total_bytes = 5376
```
The board flashing is chaotic and doesn't works all time. I managed to flash with target flash2 and flash3 too but don't know exactly why it's so random.

benjaminSchilling33, 2024-07-25:
thanks! ´flash´ did not work at first, then i tried ´flash2´ which was successful. ´flash´ and ´flash3´ are not very reliable and sometimes fail, exatly as you describe. Also sometimes, the device is not rebooted after flashing or something leaving it in ´not blinking state´

Martoni, 2024-07-26:
I wrote an [issue](https://github.com/efabless/clear/issues/35) for this chaotic flash.